LumaTeam/Luma3DS #743: Softlock upon starting Input Redir. immediately upon boot

LinkSoraZelda, 2017-07-23:
**System model:** New Nintendo 3DS XL

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):** 11.4.0-37J SysNAND

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):** boot9strap 1.2

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:** 8.1 stable

[e.g. 8.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4)

Splash: ( Before)

PIN lock: (4)

New 3DS CPU: (Clock)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
In short,  opening the Rosalina menu seconds into booting the home menu, then proceeding to open Input Redirection, will cause the console to softlock upon exit, requiring a reboot. Of course, this could always be remedied by just waiting, but it's still an issue you should probably know about. If there were a way to detect a fresh boot of the home menu, Rosalina menu requests should be ignored for a few seconds. 

The error by Rosalina is 0x09401bfe, a standard timeout error.

**Steps to reproduce:**

1. Boot up the system and open Rosalina within 5 seconds of the HOME Menu appearing.

2. Start Input Redirection, and exit

**Dump file:**

Hikari-chin, 2017-07-23:
Confirmed this is an issue, at least partially. I didn't lock up, but doing this caused most apps/folders and the battery/wifi/etc. bar to never appear.

fox8091, 2017-07-24:
Also reproduced (with debugger as well), and when in the state you were in @Hikari-chin, opening and closing Rosalina again softlocked the system as well. Once in this state, any attempts to open InputRedirection result in error code 0xFFFFFFFF, though debugger still returns 0x09401BFE.

TuxSH, 2017-07-28:
Old issue and I don't know how to fix.

closed 2017-07-28 by TuxSH

Xamanthas, 2017-07-28:
Same issue and I was able to get it to crash

[crash_dump_00000007.zip](https://github.com/AuroraWright/Luma3DS/files/1182901/crash_dump_00000007.zip)