LumaTeam/Luma3DS #746: Pokémon moon randomized crashes cfw

andy5861, 2017-07-25:
Every time I try to launch ,y Pokémon moon randomized CIA file I installed it crashes my cow im running the latest luma and boot 9 strap but it keeps crashing

jupitteer, 2017-07-25:
Github is not for technical support. I recommend going to the 3ds homebrew discord.

andy5861, 2017-07-25:
I could leave the dmp file it leaves on my sad card

HugaShire, 2017-07-25:
@Andy5861 Show a pic of the 3DS and upload the dump file. Also you should fill out the github issue template

andy5861, 2017-07-25:
[image]

andy5861, 2017-07-25:
https://www.dropbox.com/sh/ayiyepqrhw478rn/AAAuIEuYUQFQc5fvlrKrAt6Ma?dl=0 my error dumps are in there

jupitteer, 2017-07-26:
Invalid CIA, possibly?

andy5861, 2017-07-26:
i made the cia from my cart

HugaShire, 2017-07-26:
Redump it. The error looks like a corrupted file (.app)

andy5861, 2017-07-26:
alright could it be pk3ds corrupting it?

andy5861, 2017-07-26:
i fixed it

jupitteer, 2017-07-26:
close the issue than

closed 2017-07-26 by andy5861