LumaTeam/Luma3DS #747: ARM 11 exception error core 1 (SWAP DOODLE BOOT)

edgelord9000, 2017-07-25:
NEW 3DS
SysNAND 11.5.0-38E
Sighax 1.2
Luma 3ds 8.1 stable release

Default EmuNAND: (-)

Screen brightness: (4)

Splash: (After)

PIN lock: (-)

New 3DS CPU: (-)

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

When I boot up swap doodle I get a ARM 11 exception error
[crash_dump_00000000.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1174533/crash_dump_00000000.dmp.zip)

Steps to Reproduce:
I literally just boot it up

jupitteer, 2017-07-25:
Do you have doodlebomb? If so, it is not compatible with rosalina. Rosalina does not support *hax payloads.

fincs, 2017-07-25:
Use Rosalina's native 3dsx loading feature instead of trying to launch *hax.

closed 2017-07-25 by fincs

edgelord9000, 2017-07-25:
I had Doodle Bomb but i am not using it I just boot up the software and it crashes imidiately

edgelord9000, 2017-07-25:
I deleted the software and made a data backup, dowloaded it again form the e-shop, and now it works! Sorry to bother you, but thanks for the help annyways