LumaTeam/Luma3DS #749: ARM error when closing SD Card Data Management with the Power button

adrot, 2017-07-27:
**System model:**

[New 3DS]

**SysNAND version:**

[11.5.0-38E SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[8.1 stable]

**Luma3DS configuration/options:**

Default EmuNAND: ( 1)

Screen brightness: (4)

Splash: ( Off)

PIN lock: ( Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )
--

**Explanation of the issue:**
Accessing the SD Card Manager in any way, and then short-pressing the POWER button delivers an arm11 error
Error does not occur when pressing the back button to go back to the other settings

**Steps to reproduce:**

1. Go to System Settings, then Data Management, then SD Card Data Management

2. Short - Press the POWER button

**Dump file:**

[crash_dump_00000006.zip](https://github.com/AuroraWright/Luma3DS/files/1181482/crash_dump_00000006.zip)

deadair3210, 2017-07-28:
I believe this is normal and happens with a non-modded system, the only difference is that menu just silently restarts instead of showing an exception.

adrot, 2017-07-30:
Yeah, but the system in this case is modded. The first time it ever happened after updating the Luma and boot9strap

ValliantStorme, 2017-07-30:
Luma3DS Rosalina hooks the error handler, and makes easily parsed crash dumps if something happens (even in situations where an unmodified 3DS would silently ignore an error). 

It's probably intended behavior on Nintendo's part to error out of MicroSD Management, as the built-in error handler ignores it.

jupitteer, 2017-08-02:
This isn't a problem with Luma. All luma does is show the error instead of silently crashing.

mackxhunter, 2017-08-06:
I need help will this work on 11.5  now

adrot, 2017-08-06:
It seems that the Sleep Mode screen will appear instead of that annoying error if you turn on ''Disable ARM11 exception handlers''.

ValliantStorme, 2017-08-07:
@adrot Yes, but turning that option on automatically makes you ineligible to submit an issue. This isn't an issue in Luma3DS, rather it's a problem with the way Ninty wrote Data Management.

TuxSH, 2017-08-07:
Old bug, Nintendo's error here.

You could disable exception handlers (but then can't report here), or you know, just use better software such as ftpd.

closed 2017-08-07 by TuxSH