LumaTeam/Luma3DS #752: [Compilation] Current and future features for Rosalina (All Rosalina feature requests go here!)

Hikari-chin, 2017-07-30:
This is a continuation of #619. 

With the release of Luma3DS 8.0 and the custom sysmodule Rosalina, we have been getting a lot of issues concerning Rosalina's current/planned features, as well as requests for new features. We've created this issue to consolidate these requests; **if you want to request a feature for Rosalina, please do so here**. If a feature has already been requested, please do not comment about your support or dislike of said feature; instead, use Github's "reaction" system to give it a thumbs up, thumbs down, like, etc. Please note that any arguing, bickering, fluff, and/or drama will be removed without notice.

The status of feature requests can be viewed [here](https://docs.google.com/spreadsheets/d/1aQNSh173QtPeNMloFfyVbPEeWxa4Dj1vlxkO8Rg8T4g/edit?usp=sharing).

icefire82G, 2017-07-30:
Alrighty, double checked the spreadsheet, so now I'm just gonna post this here since it doesn't seem to have been noticed:

What would be convenient is a feature that could prevent the device entering sleep mode when the clamshell (not the right word but close enough) is closed.

I'm currently setting up a system where I stream games from my 3DS to my PC (using NTR, in the future a capture card), and then using Input Redirection in order to play them with my Xbox controller. As convenient as it is to have the device open to monitor lag, my current lack of space makes it an issue, so just having the device closed and out of the way would be convenient. Probably not the best solution, but convenient for the time being.

TheMCNerd2017, 2017-07-30:
How about the ability to view more details about the running processes in the process list, like the amount of CPU and RAM they are using?

Dimensional, 2017-07-31:
I certainly would love to see a GW-like debugger/cheat system. Easier to use than... NTR! *lightning crackles*

jupitteer, 2017-07-31:
Would a RTC modifier such as the recent homebrew be possible to implement into Rosalina?

WubTheGame, 2017-07-31:
Can the date and time be displayed in the Rosalina menu alongside the battery percentage?

Valeri0p, 2017-07-31:
Sorry for repeating something already said in the other issue, but why there isn't any mention to the plug in system in the requests chart?

Valeri0p, 2017-08-01:
Ok, save slots are rejected from what I can see in the list, but what about save states (aka real time saves, like NTR)?
They are pretty much a dump/injection of the current process memory, right? 

Also, thanks for deleting a legit question without giving any answer... I'm pretty sure I didn't see any mention to this "dynamic loader" in the list yesterday...only gmail knows what happens when a bad comment pops up...
Please only delete this part if you hate me.

vaguerant, 2017-08-01:
Back in #619, @StateOfException suggested persistent per-game Clock+L2 settings, as a feature ala locale settings which can be dropped in ```/luma/titles/```. This feature wasn't added to the issue post or spreadsheet, so as far as I know it hasn't been rejected, per se. It would be ideal for the handful of games which don't handle the New 3DS features (from memory, I think L2 crashes Sonic Racing, clock crashes Ridge Racer and Hana to Ikimono Rittai Zukan; I'm sure there are more examples).

obscurename, 2017-08-01:
How about a part in Rosalina that lets you take text-only notes? Because the game notes application isn't enough for long notes.

And also, an option to disable Rosalina in Luma settings if some people do not need to use it, so save up some RAM?

TheSitX, 2017-08-02:
1. possibility of load plugin like NTR...included in rosalina menu
2. 3d and volume switcher for broken's buttons
3. Select who game patch in /luma/titles/ folder ....are active or not
possibility to record video directly on console ...like screenshot' function

Hikari-chin, 2017-08-02:
So uh, guys. Please actually read the body of the main post and view the spreadsheet before commenting. Like I said, redundant/repeated feature requests (**especially for things that have already been explicitly rejected on the spreadsheet**) will just be deleted.

DMSalesman, 2017-08-03:
Would it be feasible to force the 3DS to connect to an access point that lacks access to the Internet? At the moment, if a network does not provide Internet access, the 3DS ignores it.

I confirm that this feature has not been mentioned anywhere on the spreadsheet.

LiquidFenrir, 2017-08-03:
There is already a tool that allows that, https://github.com/Lectem/FakeMii

DMSalesman, 2017-08-03:
I wasn't aware of that. Please disregard my proposal, as I'll look into it.

lavendertheghost, 2017-08-03:
Right, as a movement from my previous post.
Improvement: A facelift for Rosalina and the boot/payload menu
Reason: The menus seem bland and can be an eye sore, however, they are practical, the aim would be to make them more appealing and not remove any functionality.

WubTheGame, 2017-08-03:
New Idea: Is there a chance there could be a "System Information" tab in the Rosalina menu? it could display things like the model, ip, whether or not the system is connected to the internet, and if so, which access point, available space on the NAND and SD Card, etc.

fox8091, 2017-08-04:
@WubTheGame More appropriate in a homebrew, like 3DSIdent.

DMSalesman, 2017-08-04:
@LiquidFenrir I actually looked into FakeMii and, seeing that it does require both node.js and a computer, decided that it would be quite the effort for a casual user to adoperate, and hence I renew my above suggestion.

suzsuzuki, 2017-08-04:
Idea: Allow "install" Rosalina in multiple games/apps and choose a 3DSX to load for each game/app

mrissaoussama, 2017-08-04:
would it be possible to implement "Title restart" to reopen the currently running title?

MHHuk, 2017-08-05:
I would like to get .gif files working as luma's splash screen.
Once you convert a .gif file into a .bin file using a converter you can't use them as luma's splash screen because they don't run.
Hope to get this some day.
Thanks to everyone for reading this.

jupitteer, 2017-08-05:
@MHHuk If you really want something like that for a splash screen, use BootCtr9.

cheatfreak47, 2017-08-05:
idea for miscellaneous feature: automatic "pixel perfect" mode enable for VC, GBA, and DS mode, when enabled you won't have to hold select or start to achieve the effect- just a convenience thing- would be quite nice.

jupitteer, 2017-08-05:
Eh, I'm fine with using twl/gba games upscaled.

WubTheGame, 2017-08-05:
For convenience, it would be awesome to be able to quick close software with just Home + X.

SvenDaHacker64, 2017-08-07:
Idea: Adjust opacity/alpha level for menu so we can see what's on bottom screen. Useful for confirming scfeenshot.

MHHuk, 2017-08-07:
@jupitteer  BootCtr9 is so bugued nowadays and makeanim.bin wich is use to create your own animations does no longer works so... you can use it but it isn't as good as it was. I have to say that only top screen is supported.

dubbz82, 2017-08-08:
How about an option to control which button exits Rosalina?  This causes issues with FBI if installing a CIA and you press the B button to close out of Rosalina, as it will pick up within FBI and cancel the install.

onmyouji, 2017-08-08:
@dubbz82 They're planning to add input delay on exit to prevent accidental presses. It will solve your problem. See the docs on the first post.

alpuy, 2017-08-08:
Right now we can see all the processes that are currently running, it could be good to have a way to kill a process.

ThunderRemix, 2017-08-09:
Are we allowed to add on to our previous suggestions? For "Gateway-like memory "debugger"", I would like to replace the "?" in the "Why?" column with Convenience or Easier Debugging. Not only will it give a lot of power to Rosalina for quick debugging or memory edits that NTR CFW cannot do, but it also can get people off of Gateway. Just want to say this because I noticed that all columns that have a "?" have been rejected thus far.

GerbilSoft, 2017-08-09:
Leaving this here for reference:

"Shortcut options" to disable and reset the menu key combo.
* "Disable" would set it to 0xFFFFFFFF.
* "Reset" would set it to L+Down+Select.

Probably not really worth the effort to implement, though.

alpuy, 2017-08-10:
Uninstall the title that is currently selected in the home screen (results in a faster and more convenient way to remove titles).

CodyMKW, 2017-08-10:
Suggestion: Maybe can add Top,Bottom, and Both options to Rosalina before it takes a screenshot that would be nice for some who only want a screenshot of one of the screens

jupitteer, 2017-08-10:
Would it be possible to play a music file from the sd card while a game is running?

WubTheGame, 2017-08-10:
So we have "Switching HBL to currently running title", but, also having the ability to hold a button to override running boot.3dsx and just launching the actual title would be convenient. This way, we wouldn't have to keep switching HBL to and from the title we want it to be on.

urherenow, 2017-08-10:
Can we please toggle game patching from within Rosalina? In the future, per-game/app toggling patches/layeredFS would also be useful in some cases.

DarkSynopsis, 2017-08-12:
Just to add to what @urherenow would be nice if we could have multiple folders in "luma/<titleid>" for example "romfs1", "romfs2" etc and withing Rosalina you could select which one to make the "active" romfs, be nice if you could pick between different modifications to run.

That might be harder to implement though, would most likely have to boot the game first to get the correct list otherwise Rosalina would have to list every title folder, a nice toggle/off per installed title would be nice for now :)

WubTheGame, 2017-08-12:
@DarkSynopsis That could easily clutter up your Luma folder. Instead, there should be a subfolder in Luma containing those folders so that you can keep everything organized, like Luma/romfs/romfs#. No clutter in Luma folder, easy to find every romfs in the romfs folder.

weirdal3333, 2017-08-14:
How about an updater for Luma3DS? you wouldn't have to install an extra CIA file in order to update to the  latest version.

Flauscheloni, 2017-08-15:
I know that this feature has already been accepted, but I don't know the details, so i just wanted to add 2 points that would be quite important to make the brightness adjustment more useful: dimming below the lowest stock setting and more granular than the 5 stock settings. If this was already being considered, sorry! I'm just really looking forward to it :)

Faazuthnar, 2017-08-15:
Idea: With the brightness controls considered, a blue light filter would be great, similar to what NTR did.

YoshiOG1, 2017-08-16:
Is there an easier way to view the RAM of a 3DS's process without having to manually make RAM dumps with the command-line GDB stub?  I just want to be able to view RAM bytes (preferably on-device) like [Nanquitas' NTR RAM explorer](https://gbatemp.net/threads/plugin-ntr-ram-explorer.419162/), but that NTR plugin freezes and is unstable.

(sorry if this sort of thing has been asked already)

TatsumaYakoshi, 2017-08-18:
Ability to change 3DS LED Notification light to a certain color, within Rosalina.

DMSalesman, 2017-08-19:
If possible, I would like a visible feedback near the battery percentage which is displayed inside Rosalina's main menu, for when the 3DS is charging. As of now, the battery percentage is displayed as if the 3DS were discharging (e.g. 50%) even if the handset is connected to a power source.

This addition might be useful in case of a busted charging LED.

3096, 2017-08-19:
Not sure if already mentioned by someone, but the ability to use more than one title as placeholder to launch more than just boot.3dsx but also other 3dsx programs would be nice.

Uumas, 2017-08-19:
Configurable input remapping would be awesome. For example:
Swap ZL and ZR with L and R
Make A + B register as R
Use the c-stick as the circle pad
I'm not sure, but it should be possible to use mostly the same code used for input redirection

CookieXD, 2017-08-25:
Is it possible to edit playcoins with the rosalina menu?
You could give yourself more playcoins without leaving the current title.

Overlay for titles with useful information like:
time, fps, Wifi connection strength, playcoins, battery percentage etc...
extremely nice would be draggable symbols on the touchscreen and movable with the circle pad.

And I had an idea for the plugin system.
Really cool would be, if it could use the same plugins as ntr, but maybe another way to save them on the SD Card (I'm not a 3ds that instantly knows which titleid is what).
Maybe the folders named the title names and the plugins named the titleid.
Or the plugins without folders named like this [TitleID]-[gamename].plg
(for example "000400000003360-ZeldaOOT3D.plg)
Or something different wich is easy for human and 3ds systems to understand.

Not a feature request directly related to Luma but I was wondering if it would be possible to create a fake nintendo update server (use another DNS to access it) in order to install a fake update which provides you CFW?
should not be impossible, right?
Would it be to difficult to actually create or could it break the way the system will handel version numbers and new real updates?

jupitteer, 2017-08-25:
@CookieXD The play coin modifier is really more suited for a standalone application.

IngeniousDefault, 2017-08-28:
Is it possible for Rosalina to change the button layout for games? For example if I press the B button the game recognizes it as the A button.

CCF100, 2017-08-29:
Is it possible to have Rosalina always display the IP address regardless weather InputRedirection/Debugger is enabled? This would be useful when I just want to use NTR to screen capture without InputRedirection.

MarioMasta64, 2017-08-29:
i second the always display ip feature.

WubTheGame, 2017-08-30:
@MarioMasta64 Then Give it a thumbs up, Not make another comment. The OP clearly states so.

legoboy0109, 2017-08-31:
Could you add a rainbow, or other color options for the notification LED like in TWLoader?

Also, a better looking gui that feels like a dashboard menu like the xbox 360's menu when you press the xbox button on the controller?

One last thing, the option to save the console's memory as a file on the sd card and loading it for VC, GBA, and DS? (Definitely sounds kind of impossible to me, but really cool if it's possible. If it's out of scope then just ignore it.)

Jimmy-Z, 2017-09-01:
Option to turn off the screen to conserve battery power while doing lengthy things like using FBI to remotely install a big CIA or using FTP to transfer large files.

We can't close the shell in that situation.

And press any button to turn it on again.

Some previous discussions: https://github.com/Steveice10/FBI/issues/373

DevShadd, 2017-09-03:
You should add in the option to dim the screen (You can dim it as much as you want, even completely turn off the screen) when 3DS goes idle for a certain amount of time (you can set a certain amount of time) which would conserve power.

RS55, 2017-09-04:
It would be good to add a booster for the sound since many games have low volume by default, that even with headphones is very low.

fox8091, 2017-09-04:
@RS55 Not possible, that's an issue with the source sound files being quiet, and you can't boost the volume beyond 100%.

DevShadd, 2017-09-05:
You can boost it over %100 just via software, it's not too hard (I hope). Just note that it might decrease battery life (because the 3DS requires more energy to get those speakers/headphones moving). And it might make the quality worse because I did hear some speakers (maybe it was headphones) making a horrible sound because they were cheap and couldn't handle that much power.

fox8091, 2017-09-05:
@DevShadd Where are you getting your information? I'm going off the fact that DSP/MCU handle this, and 100% are as high as it goes.

[304 more replies]

closed 2022-06-05 by suppai