LumaTeam/Luma3DS #753: Phoenix Wright game repeatedly crashing at the same scene.

zando95, 2017-07-30:
**System model:** N3DSXL

**SysNAND version (+emu/redNAND version if applicable):** SysNAND 11.3.0-36U

**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap

**Luma3DS version:** 8.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: ( 1)

Screen brightness: ( 4)

Splash: ( After)

PIN lock: ( Off)

New 3DS CPU: ( Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

My 3DS keeps crashing at the same scene in Ace Attorney. It's very near the end of the second case, right after Mayor Tenma says "I don't want to keep my fans waiting!" and the screen fades to black. I get this message on my screen:

`
An error occurred (ErrDisp)

Error type:       generic

Process ID:       41
Process name:     GS5
Process title ID: 0x00040000000f1400

Address:          0x003c89f4
Error code:       0xd900458b

Press any button to reboot
`

**Steps to reproduce:**

1. Go to that scene in Ace Attorney: Dual Destinies

**Dump file:**

The error message did not tell me where the dump is. I did take a video, though: https://streamable.com/c7im4

AuroraWright, 2017-07-30:
Could you extract the savedata files (just before that point) with FBI and post them here?

zando95, 2017-07-30:
Yeah I used JK's Save Manager but here it is.
[error.zip](https://github.com/AuroraWright/Luma3DS/files/1185913/error.zip)

zando95, 2017-07-30:
Welp seems like my game was corrupted. I ran the repair tool on the eShop, re-downloaded, and it works fine now.

Not a Luma issue.

closed 2017-07-30 by zando95

Hikari-chin, 2017-07-30:
Thanks for the report!