LumaTeam/Luma3DS #756: arm 9 exception, data short

MANIAK8887410, 2017-08-02:
System model:

 New 3DS

SysNAND version (+emu/redNAND version if applicable):

[11.5.0-38E SysNAND]

Entrypoint: Boot9Strap1.2

Luma3DS version: 8.1 

Luma3DS configuration/options:

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )
[crash_dump_00000007.zip](https://github.com/AuroraWright/Luma3DS/files/1193650/crash_dump_00000007.zip)

Enable game patching: ( )

Show NAND or user string in System Settings: ( X )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( X )

--

**Explanation of the issue:**

The problem i have , is that when i am on the main screen and open ntr or go to data management . It crashes with the arm9 exception data short. 

[crash_dump_00000007.zip](https://github.com/AuroraWright/Luma3DS/files/1193639/crash_dump_00000007.zip)

LiquidFenrir, 2017-08-02:
Read the information on the bottom screen in the luma settings, disabling the arm11 exception handlers "disqualifies you from submitting issues". Turn off that option, try again, and send a proper issue if needed.

Hikari-chin, 2017-08-02:
Disabled the ARM11 handlers, issue invalidated.

closed 2017-08-02 by Hikari-chin