LumaTeam/Luma3DS #759: Region Patch not working; Causing prefetch abort

icefire82G, 2017-08-05:
**System model:**

New3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

Luma 8.1-a4629e4

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: (before)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (*)

Show NAND or user string in System Settings: (*)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

When booting a European copy of Miitopia (00040000001b4f00) using Luma Region patching, it causes a prefetch abort, no matter what combination of Region strings I've tried (EUR EN, EUR EN EN 01, EUR EN EN, EUR EN UK, EUR EN GB).

The file is /luma/titles/00040000001b4f00/locale.txt like it's supposed to be, and none of the above strings that I've tested seem to prevent the crash.

**Steps to reproduce:**

1. Create proper directories for region patch (titles directory, then titleid directory)

2. Create locale.txt with only contents being "EUR EN"

3. Launch game

4. Crash

**Dump file:**

[crash_dump_00000041.zip](https://github.com/AuroraWright/Luma3DS/files/1202270/crash_dump_00000041.zip)

jupitteer, 2017-08-05:
You disabled arm11 exception handlers. That disqualifies you.

TuxSH, 2017-08-05:
Does it happen w/o region patch?

astronautlevel2, 2017-08-05:
@jupitteer I don't see where they disabled exception vectors. Also, they explain the exact issue they experience (prefetch abort)

icefire82G, 2017-08-05:
@TuxSH Yep, still crashes without the region patch.

Crash dump here: 
[crash_dump_00000045.zip](https://github.com/AuroraWright/Luma3DS/files/1202727/crash_dump_00000045.zip)

AuroraWright, 2017-08-05:
This is loader crashing and not the game itself... Does it crash with "game patching" disabled? If so, the game is probably corrupted (seed wasn't installed properly or something).

icefire82G, 2017-08-05:
Yep, it crashed when I turned off game patching.

Reinstalled the game and it worked fine after reenabling game patching and adding the proper patch.

Thanks guys!

closed 2017-08-06 by AuroraWright