LumaTeam/Luma3DS #762: Edward

mackxhunter, 2017-08-06:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

mackxhunter, 2017-08-06:
Hey this is macr now I don't get it will this work on

mackxhunter, 2017-08-06:
Hey this is Edward speaking will this work on  11.5  if Doe's just let me know.

closed 2017-08-06 by mackxhunter

james-d-elliott, 2017-08-06:
yes, works. You can find 3ds hacking help here: https://discord.gg/C29hYvh

This spot is for issues, not for questions.