LumaTeam/Luma3DS #764: Rosalina increased battery compustion

MissPotato, 2017-08-08:
**System model:** Old 3DS

**SysNAND version (+emu/redNAND version if applicable):** Don't know, so assume N/A

**Entrypoint (How/what you're using to boot Luma3DS):** Soundhax/Boot9Strap

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:** 8.1.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4)

Splash: (after)

PIN lock: (off)

New 3DS CPU: (n/a)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: (x)

Enable loading external FIRMs and modules: (x)

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Noticeable decrease in battery life when leaving Roslilina open and closing the lid. The battery won't even make it a full night with Rosalina open while with it closed about a week of idle time is my average.

samexner, 2017-08-08:
Why leave Rosalina open overnight?

Hikari-chin, 2017-08-08:
Sleep mode doesn't work when the Rosalina menu is open, known issue.

closed 2017-08-08 by Hikari-chin

MissPotato, 2017-08-08:
Oh, it didn't even link up in my head that sleep mode was what was saving battery. Sorry for the duplicate!