LumaTeam/Luma3DS #765: Issue while trying to write pin code too fast

Uumas, 2017-08-08:
**System model:**

Happens on all

**SysNAND version (+emu/redNAND version if applicable):**

Isn't relevant, 11.5

**Entrypoint (How/what you're using to boot Luma3DS):**

Isn't relevant, b9s

**Luma3DS version:**

Tested on 6.6-8.1.1

**Luma3DS configuration/options:**

Don't make a difference, anything with pin lock enabled.

**Explanation of the issue:**

If while writing pin code you push down another button before releasing the previous one it'll register one button press more than it should have.

**Steps to reproduce:**

1. Enable pin

2. Press a button used on the pin code and keep holding it

3. Press another button used in the pin code

4. Release both

5. There are now more than two characters written

jupitteer, 2017-08-09:
Don't press multiple characters down at the same time.

Mike15678, 2017-08-09:
Luma3ds pins are supposed to be entered one button at a time. You can't hold two or more at a time when entering your pin.

Uumas, 2017-08-09:
Yes, but it often happens accidentally when you're trying to type it too fast

astronautlevel2, 2017-08-09:
Then slow down

Hikari-chin, 2017-08-09:
User error, sorry.

closed 2017-08-09 by Hikari-chin

Uumas, 2017-08-10:
I understand this is kind of a user error, but as I've done a bit of coding myself (really not much) I think this could be fixed by changing code. If I have time I might dig through the Luma code to find out myself. I'd appreciate if someone could point me to the file that has the code which handles pin code button presses. Thanks.