LumaTeam/Luma3DS #766: Error code: 0xf9606c02

chelo146, 2017-08-09:
System model:

Old 3DS

SysNAND version (+emu/redNAND version if applicable):

11.3.0-36E SysNAND

Entrypoint (How/what you're using to boot Luma3DS):

Boot9Strap 1.2

Luma3DS version:

Luma 8.1.1

Luma3DS configuration/options:

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

Explanation of the issue:
After install Luma 8.0, FBI and HBL are working ok, but when the wifi is turned on, console freezes in 15-20 seconds. If the wifi is turned off, the console is working fine.

Steps to reproduce:

Start the console

Turn on wifi

Dump file:
An error occurred (ErrDisp)

Error type: generic
Process ID: 25
Process name: num
Process title ID: 0x0004012000002d02

Adress: 0x0010941a
Error code: 0xf9606c02

Press any button to reboot

jupitteer, 2017-08-09:
What's up with all of the exceptions related to turning wifi on? I've been seeing that a lot lately.

Hikari-chin, 2017-08-09:
Can't reproduce; this is likely either a system corruption issue or damaged WiFi hardware. Try CTRtransferring to a clean CTRNAND and if that doesn't fix it, it's likely your device has hardware damage.

closed 2017-08-09 by Hikari-chin

zm1868179, 2017-08-13:
System model:
Old 3DS
SysNAND version (+emu/redNAND version if applicable):
11.4.0-37U SysNAND
Entrypoint (How/what you're using to boot Luma3DS):
Boot9Strap 1.2
Luma3DS version:
Luma 8.1.1
Luma3DS configuration/options:
Default EmuNAND: ( )
Screen brightness: ( )
Splash: ( )
PIN lock: ( )
--
Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Use custom path: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: (* )
Show GBA boot screen in patched AGB_FIRM: (* )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )
--
Explanation of the issue:
3DS works fine if WIFI is off once wifi is on any attempt to access the wireless such as waiting for 20 seconds after start up or going to settings and attempting to add a wireless connect causes it to crash. 3DS has been unused for weeks only changed luma version to 8.1.1 even did a ctrnand transfer to 9.2 to see if there was nand corruption this did not solve the issue. This only happens on the o3ds it doesn't effect my n3ds
Steps to reproduce:
Start the console
Turn on wifi
wait a few seconds 
or 
start settings 
turn on wifi
attempt to add wireless connection
Dump file:
An error occurred (ErrDisp)
Error type: generic
Process ID: 25
Process name: num
Process title ID: 0x0004012000002d02
Adress: 0x00109466
Error code: 0xf9606c02
Press any button to reboot

sai-eron, 2018-10-12:
can someone help me to do this CTRtransfer

ValentinLaTortue, 2019-05-12:
Hello,
I have been having the exact same problem, which is very fustrating considering I bought the 3ds to play multiplayer games.

How do we CTRtransfer to a clear CTRNAND? Is it by following the tutorial from [https://3ds.hacks.guide/ctrtransfer.html](url)?

Is there an other way to fix that problem?
What to do if it is an hardware issue? I have no problem taking my 3ds appart.

I want to be sure of what I'm doing before I do anything irreversible.

Victorh2002, 2019-12-24:
I have the same problem and and this didnt solved my problem.