LumaTeam/Luma3DS #769: I don't find boot.firm

Meyleen, 2017-08-10:
Hello I follow a tutorial for having free game and i must install lumia

In this tutorial he says "install the boot.firm from lumia .7z in your root of your sd card"

In download all files of lumia (the last version) but i havn't this files...

I tested to continuate the tutorial without this files but now my 3DS don't working... (after start the homebrew laucher, the screen beacame black and now my 3DS does not light up

SO where can i find this files??? I need it, absollutly!

astronautlevel2, 2017-08-10:
This isn't a troubleshooting forum, can you try on The [Nintendo Homebrew](https://discord.gg/C29hYvh) discord?

closed 2017-08-10 by Stary2001

MasterGamingYT, 2019-04-06:
Ask Homebrew discord for help, also we don't support piracy here sorry!