LumaTeam/Luma3DS #772: Bug in Luma3DS 8.1 / 8.1.1

Winchoos, 2017-08-10:
Greetings.
Found a rare bug in Luma3DS for some users on 8.1 / 8.1.1. Standby mode: set to sleep, after a certain amount of time open the console - the screen does not respond, the blue indication. It is treated by pressing the "Power" button until the console is completely turned off. Please extend the 7.1 branch of Legacy. Thank you for your attention, all the world.
PS I apologize for my English, not my native language.

TuxSH, 2017-08-10:
Hi,

I'm not able to reproduce this issue myself.

>  a certain amount of time 

How long ?

> Please extend the 7.1 branch of Legacy.

I don't think it is a Luma 8.x regression, and I think all CFWs are affected. To see if this is what I think about: are those users using emuNANDs, by any chance?

Hikari-chin, 2017-08-11:
Can't reproduce this on emuNAND or sysNAND no matter how many times I try or how long I wait.

closed 2017-08-11 by Hikari-chin

urherenow, 2017-08-11:
I've never had this issue on ANY CFW and play around quite a bit, without actually playing games. But I've left it in sleep mode for days at a time...