LumaTeam/Luma3DS #773: Freeshop Update/Exit Error

SauceGuy, 2017-08-10:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.1.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: 1x(x)

Screen brightness: 4x(x)

Splash: Off(x)

PIN lock: Off(x)

New 3DS CPU: Clock(x)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
8.1
I entered the Freeshop and it started updating to 3.1.4. Once the update was done, I just wanted to check to see if anything was updated and few things were so then I decided to exit the Freeshop by pressing start, like you're supposed to. Once I press start, the screen suddenly goes black and goes to an error message that shows a dump crash and several other things. It also says to press any button to turn off the system. I then proceed to go back into the Freeshop and to try and close it again but the same error message shows up. From there, I deleted the Freeshop folder located in /3ds/data/ and redownload the app using the TitleDB (arc13 fork) in FBI (I still had the app in place and did not physically delete it using Data Managment). Once that was done, I went back into the Freeshop and everything seemed fine until I pressed start again. It went to the same error message and I rebooted my system. 
7.1 Legacy Build
I proceeded to do the 7.1 Legacy build to see if that would fix it. I went into Freeshop again and pressed start but this time, the screen froze. It then took me back to the home screen saying an error occurred and that the system will restart (not the error screen I had in 8.1). I tried it again and it did the same thing. So then I deleted the Freeshop folder in /3ds/data and repeated the redownloading process from TitleDB but it still didn't work.
8.1.1
I went back up to 8.1, then to 8.1.1 in hopes that the error would fix itself. I went into the Freeshop and pressed start like usual and it went to the black error screen in 8.1...again. I got a little impatient so I deleted the Freeshop folder like usual but this time I went to Data Managment and physically deleted it. I went and reinstalled the Freeshop app from scratch and went to that 3ds title keys website to get the QR code to update the Freeshop's title keys. But as soon as the reader read the QR code, the screen went back to that black error screen. So I guess the Freeshop has two ways to crash it.

**Steps to reproduce:**
8.1, 7.1 Legacy Build. 8.1.1
Method 1
1.  Have Freeshop already (not installed recently, had Freeshop for a while)

2.  Freeshop will install update 3.1.4 (depends if you already have the update or not)

3. Press Start and the screen will go to a black screen and then to an error screen (arm11) or to the home screen (7.1) which will make your system restart (make sure to have the title keys already installed)
(This will happen no matter what Luma you are on)
Method 2
8.1.1
1. Install Freeshop from scratch using FBI QR code scanner (Delete Freeshop if you already have it using Data Managment and also delete the folder inside /3ds/data/

2. Open Freeshop

3. Go to the Update tab 

4. Tick the top check box

5. Press the QR code button underneath the second check box

6. Find the 3ds title keys website and get the QR code of encTitleKeys.bin

7. Scan it with the Freeshop QR code reader and a black error screen will appear

**Dump file:**

Here are all of the crash dumps I've gotten: (some vary in size)
[Crash Dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1216556/Crash.Dumps.zip)

Hikari-chin, 2017-08-11:
We don't support piracy, sorry.

closed 2017-08-11 by Hikari-chin

DMSalesman, 2017-08-16:
@SauceGuy A crash upon exiting from that software with Start is a bug pertaining to said software, as it was introduced on the latest version of the abovementioned (i.e. nothing to do with Luma, so off-topic here).