LumaTeam/Luma3DS #777: Game Boy games abort with custom binaries

puddydoo, 2017-08-12:
**System model:** New 3DS
**SysNAND version:** 11.5.0-38U
**Entrypoint:** Boot9Strap
**Luma3DS version:** 8.1.1
**Luma3DS configuration/options:**
New 3DS CPU: Clock+L2
Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: (x)
Show NAND or user string in System Settings: (x)
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )
Disable ARM11 exception handlers: ( )

**Explanation of the issue:**

Whenever I try to use custom code.bin with Game Boy VC games, it crashes with a prefetch abort. RomFS works, but putting in a code.bin makes it crash whether a custom romFS is included or not. I have tried this with three games, Mario's Picross, Pokemon TCG, and Pokemon Amarilla (Spanish Yellow). They are all unique as Picross is an original GB game, while Amarilla has wireless support.

The crash happens whether the code.bin is from a different VC game, or whether it is the same as the one from the base title except for a single nibble value, or whether it is the exact code.bin from the base title without any difference at all. Though the game is compatible with the code.bin, it's just that the presence of a code.bin is causing it to crash. I have not tried with other VCs such as NES, SNES, Game Gear, etc.

**Steps to reproduce:**

1. Put a code.bin in the luma titleid folder of a Virtual Console game (Game Boy). A code.bin extracted from that game should do.
2. Open the VC game

**Dump file:** [crash_dump_00000020.zip](https://github.com/AuroraWright/Luma3DS/files/1219469/crash_dump_00000020.zip) (From starting Pokemon Amarilla when its title folder contains nothing but its own code.bin)

AuroraWright, 2017-08-12:
How did you get the code.bin? It needs to be decompressed

puddydoo, 2017-08-12:
I did not know that the code.bin needs to be decompressed. With code.bin decompressed, each one worked (or broke the game) as it should.

closed 2017-08-12 by puddydoo