LumaTeam/Luma3DS #778: Cursor position in rosalina

Fabbbrrr, 2017-08-12:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.1.1 stable

**Explanation of the issue:**

In Rosalina, the selected option position is not stored (or recalculated) on every menu/sub menu. Instead, the last position is used, which produces weird positioning of the cursor when going back in the menu.

**Steps to reproduce:**

1. Open rosalina (cursor at the first option)
2. Go to Misc options.
3. Press DOWN 3 times. (cursor at the Save state option, position 4/4)
4. Press B.
5. In the main rosalina menu, the cursor will be at position 4/10, instead of the last selected (Misc Opt - 7) or the first one.

WubTheGame, 2017-08-13:
I can confirm this working on 11.5 with OLD3DS B9S 1.2 and Luma 8.1.1

ValliantStorme, 2017-08-14:
I can confirm this is a bug that occurs on Luma 8.1.1 and the current hourly (9c6b5409)

Derek1199, 2017-08-14:
Haven't noticed anything in rosalina tbh

commit ca2622af (2017-08-16): Fix #778
closed 2017-08-16 by TuxSH