LumaTeam/Luma3DS #779: [Request] Boot into a title with TitleID in "tid" file, if exists

Robz8, 2017-08-12:
I thought of another way an app like TWLoader can be booted after exiting TWL_FIRM.
Luma should boot into a title with TitleID specified in a file called "tid" if the file exists, instead of the HOME Menu, then the "tid" file can be deleted after it's read.
For example, if an NTR/TWL-mode homebrew app wanted to boot an app like Face Raiders, the app can write it's TitleID in the "tid" file, then attempt to reboot into HOME Menu, but if the "tid" file is found, it'll boot Face Raiders instead of the HOME Menu.
Also, "tid" file can be skipped and deleted, if B button is held.

JoshuaDoes, 2017-08-12:
I have a better idea about the buttons: Instead of B, use L to skip it and delete it. And use R to use it but not delete it.

Robz8, 2017-08-12:
Yeah, that's better.

Faazuthnar, 2017-08-16:
This could be a way to skip the Home menu on boot and go straight to a game, or boot whatever is on the flashcard (how the DS used to), or if you want to, go straight to the Homebrew Launcher on boot.

fincs, 2017-08-16:
I think that having to rely on writing and reading a file potentially on every boot is not a very elegant solution & it is not good for the SD card. I would try to find another place to place the tid/mediatype in that persists after a reboot (but not a shutdown)

Jan200101, 2017-08-16:
write it to nand

fincs, 2017-08-16:
It's not good for NAND either.

Robz8, 2017-08-17:
Why not? Isn't NAND written to, whenever the 3DS is used/booted?

joel16, 2017-08-17:
+1 with fincs here, please no.

TuxSH, 2017-08-17:
At the very least it should be an option & implemented for 3DS titles as well. The problem is that you need to boot into a different memory mode on O3DS...

fox8091, 2017-08-17:
Perhaps ITCM/DTCM? However, that can't be accessed from ARM11... I don't know.

jupitteer, 2017-08-18:
Seems like an arm11 version of the boot9strap build that had bootonce support...

SirNapkin1334, 2018-01-17:
Since I'm assuming this would only be used in a reboot and not coldboot, why not read it into RAM from some other place, and manipulate it from there? Perhaps it can be stored on the TWLNAND partition if it is used after exiting TWL_FIRM.

profi200, 2018-01-18:
Actually writing that to ITCM should be fine if you find a safe location. Don't unnecessarily write to the SD card. Small files like that which are often written to are not good for the flash.

JoshuaDoes, 2018-08-02:
Yeah, seems like writing to and reading from NAND on every reboot a little more than it already does would just wear out the NAND quicker than usual. Hopefully there are super safe locations in RAM where something like this could be stored, which would help with the idea of it only being a reboot thing and not a coldboot thing.

A little bit of research shows that some areas of memory aren't wiped on reboots (but obviously are lost after a shutdown and are empty on a coldboot), so maybe we could hide it in one of those regions?

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH