LumaTeam/Luma3DS #780: ARM9 Crash Error

Xercanion, 2017-08-12:
**Luma3DS configuration/options:**

Default EmuNAND: (No )

Screen brightness: (4 )

Splash: None

Enable game patching: (No )

Show NAND or user string in System Settings: (Yes )

Show GBA boot screen in patched AGB_FIRM: (No )

Patch ARM9 access: ( No)

Set developer UNITINFO: (No )

Disable ARM11 exception handlers: ( No)

--

**Explanation of the issue:**
Every time when i run Decrypt9 FBI or the Power Button i get a ARM9 Crash error afterwards. This started to happen frequently after i updated luma 3ds from 7.1 to 8.1 but i can stil use my 3ds fine.

**Steps to reproduce:**

1. Updated Luma 3ds from 7.1 to 8.1 

2.Run FBi or Decypt9 and you will get the crash error 

**Dump file:**
https://www.dropbox.com/s/bbq9c05gh1dsm7r/crash_dump_00000004.dmp?dl=0

Robz8, 2017-08-12:
Are they both up to date?

Xercanion, 2017-08-12:
yes they are

Xercanion, 2017-08-12:
do you want a screenshot of my sd card root as well?

Derek1199, 2017-08-12:
Cannot reproduce this. Be more descriptive. You should be using luma 8.1.1
instead of 8.1 as luma 8.1.1 is the latest. Try using that.

Xercanion, 2017-08-12:
LumaUpdater states the current version is on 8.1.1

Derek1199, 2017-08-12:
Okay. Is that what it says in luma config as well? You can access that by
holding select on boot. Also is your 3ds an n3ds or o3ds?(n3ds is new 3ds.
o3ds is old 3ds.)

Xercanion, 2017-08-12:
yes version 8.1.1 is also mentioned on the configuration menu of luma  im currently on the n3ds firmware 11.5 my previous firmware version was 11.4

Derek1199, 2017-08-12:
Okay. You said you have FBI and decrypt9 up to date. However did you
actually check them? When did you last update them?

Xercanion, 2017-08-12:
i updated it two days ago

Xercanion, 2017-08-12:
heres a video link on how it happens sorry for the background noises 

https://www.dropbox.com/s/tp8vueo1e4381a2/20170812_142837.mp4?dl=0

Derek1199, 2017-08-12:
Okay. Well, you might've done something to your firm0 or firm1 partition I
guess. I cannot reproduce this. I can give you compiled builds of
boot9strap and luma if you want. Latest commit of both. Which would have
latest code changes. But it'll be a few hours before I get home. Doing that
should fix your issue I hope.

Xercanion, 2017-08-12:
Thanks Derek ☺

Xercanion, 2017-08-12:
should i i dont maybe try to reinstall boot9strap from scratch to see if that resolves the issue?

Derek1199, 2017-08-12:
Yeah but instead use the ones I put here later. It will have latest code
changes. So any possible issues should be fixed for you after.

Xercanion, 2017-08-12:
ok and by the way this is what my sd card root looks like :
https://www.dropbox.com/s/o15qc9bbjryzgnj/20170812_143939.jpg?dl=0

This one inside of boot9strap folder:
https://www.dropbox.com/s/jbr194nhorixmm2/20170812_143956.jpg?dl=0

Derek1199, 2017-08-12:
Okay. I'll give you compiled builds later. Around 2-5 hours from now I
think. Not home right now. They should fix your issue hopefully.

Hikari-chin, 2017-08-12:
You can't launch Decrypt9 from the Home Menu or Homebrew Menu without an ARM9 exploit or other screwery, it's an ARM9 payload (and has been discontinued anyway). As for FBI, I doubt it's up to date, both of these crashes could easily be caused by them attempting to gain ARM9 permissions through out-of-date incompatible exploits. So, make sure you have: 

* The [latest version of the Homebrew Menu](https://github.com/fincs/new-hbmenu/releases)
* The [latest version of [FBI](https://github.com/Steveice10/FBI/releases)

If you have anymore questions, please ask them [here](https://discord.gg/C29hYvh).

closed 2017-08-12 by Hikari-chin