LumaTeam/Luma3DS #785: Wifi on causes crash

zm1868179, 2017-08-14:
System model:
Old 3DS
SysNAND version (+emu/redNAND version if applicable):
11.4.0-37U SysNAND
Entrypoint (How/what you're using to boot Luma3DS):
Boot9Strap 1.2
Luma3DS version:
Luma 8.1.1
Luma3DS configuration/options:
Default EmuNAND: ( )
Screen brightness: ( )
Splash: ( )
PIN lock: ( )

Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Use custom path: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: (* )
Show GBA boot screen in patched AGB_FIRM: (* )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )

Explanation of the issue:
3DS works fine if WIFI is off once wifi is on any attempt to access the wireless such as waiting for 20 seconds after start up or going to settings and attempting to add a wireless connect causes it to crash. 3DS has been unused for weeks only changed luma version to 8.1.1 even did a ctrnand transfer to 9.2 to see if there was nand corruption this did not solve the issue. This only happens on the o3ds it doesn't effect my n3ds
Steps to reproduce:
Start the console
Turn on wifi
wait a few seconds
or
start settings
turn on wifi
attempt to add wireless connection
Dump file:
An error occurred (ErrDisp)
Error type: generic
Process ID: 25
Process name: num
Process title ID: 0x0004012000002d02
Adress: 0x00109466
Error code: 0xf9606c02
Press any button to reboot

Hikari-chin, 2017-08-14:
Reposting your comment from a closed issue about this very topic as a new issue won't get you anywhere...

closed 2017-08-14 by Hikari-chin

luisho24, 2017-10-21:
I am having the same issue. The fact that it cannot be reproduced does not mean that it cannot be resolved with a logic solution. everything cannot be reproduced there are always variables.

arikazen, 2017-11-22:
I have the same problem. began today. I dont know what i do. Please help.

zm1868179, 2017-11-22:
I think this may be fixed to failed wifi module I recently replaced the module and the error went away. Would be nice to have a different error message or atleast not cause a system crash due to a failed module