LumaTeam/Luma3DS #795: ARM11 EXCEPTION WHEN OPEN 3DS DOWNLOAD HB TITLE

Juanito9001, 2017-08-17:
I do the steps to open the hb launcher in 3ds download, but when i ooen it again, appears ARM11 exception, a bunch of numbers and "press any key to shut down" what can be the problem?

_Sent from my Samsung SM-G610M using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

Derek1199, 2017-08-17:
What version of Luma do you have?

Juanito9001, 2017-08-17:
Luma 8.1.1
Boot9strap 1.2
Old3ds 11.5

_Sent from my Samsung SM-G610M using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

Juanito9001, 2017-08-17:
I fix it, the problem was i don't doenload the new version of the hblauncher that supports rosalina

_Sent from my Samsung SM-G610M using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

closed 2017-08-17 by Hikari-chin