LumaTeam/Luma3DS #798: Random crashes

LennyMcLennington, 2017-08-18:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

B9S

**Luma3DS version:**

8.1.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Before)

PIN lock: (4)

New 3DS CPU: (N/A)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Random crashes while the 3DS is on the nand (not in the chain loader / config / payloads).

**Steps to reproduce:**

1. Turn 3DS on

2. (Optional) Open a game or app.

3. Wait

4. Crash

**Dump file:**

[relevant dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1234580/relevant.dumps.zip)
1st dump is Super Mario Maker.
2nd is Sm4sh.
3rd is on home menu.

LennyMcLennington, 2017-08-18:
Also I'm banned from that support server

AuroraWright, 2017-08-18:
Does it happen if you turn wifi off?

LennyMcLennington, 2017-08-18:
I will check

LennyMcLennington, 2017-08-18:
Nothing so far with WiFi on, my brother knows more than me about the issue bc it's his 3ds but he's out rn

LennyMcLennington, 2017-08-18:
I'm timing it with WiFi on then I'll try with WiFi off

LennyMcLennington, 2017-08-18:
No crashes yet maybe just extreme bad luck

closed 2017-08-18 by LennyMcLennington

AuroraWright, 2017-08-18:
Probably wifi card related, given that nwm is always the one crashing