LumaTeam/Luma3DS #800: Exception error

sterob, 2017-08-19:
**System model:**

 New 3DS 

**SysNAND version (+emu/redNAND version if applicable):**

9.2.0-20U SysNAND, 11.0.2-37U EmuNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.1.1

**Luma3DS configuration/options:**

 Default setting

**Explanation of the issue:**

So i was moving from menuhack to this B9S. 

I launched soundhax and somehow i keep reboot me back to the bootmenu.

Then i read, soundhax is only one of the way to go into homebrew launcher. So i go strait to the homebrew launch (before booting into sysnand). 

Then i think i run safehax before udsploit. After a screen about backing up FIRM, i got into Luma configuration screen, press enter my 3ds crash with this (http://i.imgur.com/2khXuWV.jpg). Poweroff and restart show the same thing. Strangely enough, i was able to boot into emunand by holding L. 

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1236340/crash_dump_00000000.zip)

sterob, 2017-08-19:
Nevermind, clearing Home Menu extdata seem worked.

closed 2017-08-19 by sterob

manu-03, 2017-08-19:
Dump your EmuNAND with GodMode9, restore it to SysNAND and delete the EmuNAND to save space in your SD card.