LumaTeam/Luma3DS #803: Get an error when playing games

almightypacman, 2017-08-20:
**System model:**

 Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38E

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

8.1.1

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Hi so I recently followed the guide to get Luma and everything went fine but when I went to play omega ruby(game cartridge) in certain areas of the game it popped up with an error on the bottom screen and it forced me to reboot. On the screen with the error it says process ID then a number depending on what I was doing on the game this number would change.

**Steps to reproduce:**

1.

2.

**Dump file:**

almightypacman, 2017-08-20:
[image]

erric1, 2017-08-20:
I met this  problem today also, i didn't do anything except playing games.

erric1, 2017-08-20:
And my error code is the same as yours

AuroraWright, 2017-08-20:
Does it happen with this? https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-3a4868c.zip

erric1, 2017-08-20:
my boot is the latest 8.1, and to solve the problem ,i replace it with 8.1.1, crash still happens.

AuroraWright, 2017-08-20:
Try the boot.firm I just sent you

erric1, 2017-08-20:
strangely, when i play mhxx online, in the quest never crashes, if finished crash happens sometimes.i will try it now,thanks

TuxSH, 2017-08-21:
The error code shows that your installation of the game (or its update data) is partially corrupted.

closed 2017-08-21 by TuxSH