LumaTeam/Luma3DS #804: [LayeredFS] Arm9

Gnarmagon, 2017-08-21:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

8.1.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (Off )

PIN lock: (Off )

New 3DS CPU: (Off )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: (x )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I get a Crash when I start The Binding of Isaac Rebirth (EUR) with the Title ID:
000400000f700900

**Steps to reproduce:**

1. Copy the luma Folder from the LayeredFS Zip onto the SDCard

2.  Start The Binding of Isaac

**Dump file:**
[crash_dump_00000002.zip](https://github.com/AuroraWright/Luma3DS/files/1238254/crash_dump_00000002.zip)

**LayeredFS File:**

AuroraWright, 2017-08-21:
Note that you can't add files which didn't exist in the original romfs, is that what you're doing here?

Gnarmagon, 2017-08-22:
oh okay :( that makes sense -.-

closed 2017-08-22 by Gnarmagon