LumaTeam/Luma3DS #806: R4i Gold 3DS Deluxe compatibility with v4.1-v4.5 Emunand

Buzbee, 2017-08-21:
**System model:**
Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**
11.5.0-38U SysNAND, 4.5.0-10U EmuNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap/Sighax

**Luma3DS version:**
8.1.1 stable 

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
MSET exploit does not work with R4i Gold 3DS Deluxe' Launcher.dat (but MSET exploits works fine with Gateway's Launcher.dat) with V4.1-v 4.5 Emunand.

R4i Gold 3DS Deluxe Edition:
R4i Gold 3DS Wood (blue card with DS games on it)
R4i Gold 3DS (yellow card with 3DS games on it)

Yellow card compatibility is for v4.1-4.5 with MSET exploit. Blue card compatibility includes v11.5.

**Steps to reproduce:**

1.  Copy 4.0B2 launcher.dat to 3DS SD card root directory (download from http://r4ids.cn/r4i-download-e.htm )

2. Insert blue card and launch R4i Wood firmware in 4.1-4.5 Emunand by pressing A on the Konami Digital Entertainment game cartridge icon.

2. Press Start button.  Tap 3DS Setup then Press B.

3. Press Home Button then tap OK. Remove Blue cartridge and insert Yellow cartridge.

4. Go to settings->Other Settings-> Profile->Nintendo DS Profile

5. Black screen occurs then freezes (properly launches Gateway Menu if you replace it with Gateway's launcher.dat). It is suppose to reload the Home Screen and display a 3DS rom game cartridge icon like it does in v4.1-v4.5 Sysnand.

**Dump file:**

AuroraWright, 2017-08-21:
Luma changes the memory layout and this probably breaks those exploits. Can't be fixed

closed 2017-08-21 by AuroraWright

TuxSH, 2017-08-21:
and if it isn't the problem we don't have to deal with something that isn't a luma problem.