LumaTeam/Luma3DS #807: NTR Cheat Plugin crashes & various other crashes.

MerchDoesGaming, 2017-08-21:
**System model**

[Old 3DS]

**SysNAND version**

[e.g. 11.5.0-38U SysNAND]

**Entrypoint**

[e.g. Boot9Strap]

**Luma3DS version:**

[8.1.1 (Stable)]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Before)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Whenever I play a game like Shadows of Valentia, & try to do something you can't do in the stock game, like screenshot using Rosalina, or enable NTR Cheats, it always finds a way to crash. This goes the same for various other apps that I assume access the NAND. I've had many other crashes before that, but I'm not sure how to decipher which one is which. I've linked all of my crash dumps in the issue, just in case there's something you may need from this.

**Steps to reproduce:**

1. Boot up 3DS; splash screen plays out.

2. Boot up any app:

- Shadows of Valentia

   1. Boot up SoV.

   2. Play through the game normally; occasionally take screenshots.

   3. Game crashes after randomly taking a screenshot.

- BootNTRSelector

   1. Boot up BootNTRSelector.

   2. Choose 3.6.

   3. Launch game (whether it be Shadows of Valentia, Smash, Mario Kart, Mario Maker, or anything that uses Select to access the cheat menu).

   4. Apply cheats.

   5. Either a toss up of cheats working, & an occasional crash, or the system crashes immediately after applying cheats.

**Dump file:**
[arm11.zip](https://github.com/AuroraWright/Luma3DS/files/1240010/arm11.zip)

Hikari-chin, 2017-08-22:
Do the crashes ever occur *without* launching NTR/having it running?

MerchDoesGaming, 2017-08-22:
Yes. Apparently when taking screenshots with Rosalina, that happens. There's also some instances where FBI seems to crash when installing a .cia that has to deal with the system. Also testing out some other shenanigans, like n3DS exclusives, & microSD Management seems to crash when disconnecting the system from your PC.

MerchDoesGaming, 2017-08-22:
As far as I know, the microSD isn't just o3DS exclusive crashing; it slso happens on n3DS.

deadair3210, 2017-08-23:
Yeah, N3DS XL running latest here, microSD is just buggy in general, either refuses to connect or outright crashes. That's why most people just use FTP

Dmestral, 2017-08-24:
[image]
hi I'm trying to run the mhxx patch with BootNtr Selector Mode3, and crash after Nintendo 3ds logo, but start the game without patch runs normal. Any way to make run with the patch?

jupitteer, 2017-08-26:
Never happens for me. Screenshotting nor ntr has never failed for me.

MerchDoesGaming, 2017-08-26:
@Hikari-chin Yes. I don't know what you mean by "similar", because that seems broad, but whenever I do screenshot something using Luma3DS, it will occasionally crash the more I use the screenshot function. Other than non-related NTR crashes, I don't recall my 3DS crashing for reasons other than NTR, Rosalina, or applications that force you to reboot or restart your 3DS when a certain action happens, such as installing a raw system title .cia with no decryptions.

Hikari-chin, 2017-08-29:
I've failed to reproduce this many times over, doing exactly what you've done, so I'm going to assume this is an NTR or one of its plugins' issue, not Rosalina's. If anything to the contrary rears up in the future, I'll reopen this.

closed 2017-08-29 by Hikari-chin