LumaTeam/Luma3DS #811: Out-of-region doesn't work with Pokemon Shuffle

Clement95, 2017-08-22:
**System model:**

[New 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[11.5.0-38E SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap/Sighax 1.2]

**Luma3DS version:**

[Luma3DS 8.1,1 stable]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (1)

Splash: (Before)

PIN lock: (Off)

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: (x)

--

**Explanation of the issue:**

I'm trying to launch Pokemon Shuffle in japanese language, I used Luma Locale Switcher to set region and language of the game to japanese, but when I launch Pokemon Shuffle, it still stays on the language of my System Settings (french)

**Steps to reproduce:**

1. Enable game patching in Luma config

2. Set Pokemon Shuffle to japanese region with Luma Locale Switcher

3. Launch Pokemon Shuffle

**Dump file:**

Hikari-chin, 2017-08-23:
You have disabled your ARM11 Exception Handlers, please read the config option description. Your right to submit issues is forfeit.

closed 2017-08-23 by Hikari-chin

Clement95, 2017-08-23:
I didn't disabled it, sorry I was wrong when I wrote this issue, but nothing errors is appeared

urherenow, 2017-08-23:
tried to comment earlier, but couldn't

Clement95, 2017-08-23:
It works now, I just changed the destination to luma/titles/folder
title id/locale.txt