LumaTeam/Luma3DS #813: booting luma right after installing

DioGaia, 2017-08-23:
**System model:**

Old 3ds

**SysNAND version (+emu/redNAND version if applicable):**

9.2.0-20U sysNAND, 11.2.0-35U EmuNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

Latest 8.1.1

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (none)

PIN lock: (none)

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

This issue came up while I was following 3ds guide exactly at step where I press start to save in section III in the installin boot9strap.

Before installing boot9strap I was using rxtools.

I get exception occurred when I try to boot.

**Steps to reproduce:**

I don't think you can reproduce this

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1244036/crash_dump_00000000.zip)

Committee, 2017-08-23:
Did you put boot.firm in the root of your sd?

DioGaia, 2017-08-23:
yes, I followed every step

Derek1199, 2017-08-23:
This is a simple Fix. Go in GodMode9 and copy the boot.firm file from the root of your SD card to sysnand CTRNAND.

DioGaia, 2017-08-23:
Okay, I will try it when I can

DioGaia, 2017-08-23:
Doesn't seem to work

Derek1199, 2017-08-23:
Um. This should work. Try copying the config.bin and the payloads folder in your luma folder to Sysnand CTRNAND/rw/luma

DioGaia, 2017-08-23:
I'm currently outside so I will try it when I'm back home.

Thank you for your help

Hikari-chin, 2017-08-23:
This typically occurs due to a damaged NAND or damaged HM data. Try deleting the HM extdata, and if that doesn't work, ctrtransfer with a clean image.

closed 2017-08-23 by Hikari-chin

DioGaia, 2017-08-23:
Deleting all home menu extdata from all folders seems to have worked.

For anyone wondering go see https://3ds.guide/troubleshooting#ts_sys_b9s

Thank you derek for trying to help