LumaTeam/Luma3DS #815: "An Exception Occured" error when starting Pokemon X Cartridge on O3DS

rds12, 2017-08-23:
**System model:**

[Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[Sys 11.5.0-38E]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap/Sighax]

**Luma3DS version:**

[8.1.1 stable]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (n/A )

--

Autoboot EmuNAND: (No)

Use EmuNAND FIRM if booting with R: (No)

Enable loading external FIRMs and modules: (No)

Enable game patching: (No)

Show NAND or user string in System Settings: (Yes)

Show GBA boot screen in patched AGB_FIRM: (No)

Patch ARM9 access: (No)

Set developer UNITINFO: (No)

Disable ARM11 exception handlers: (No)

--

Hey,
thank you for your awesome work. And your amazing open-to-closed ticket ratio indicates that you may can help me :)

When starting the game "Pokemon X (EUR Version)" the game crashes immediately. Today, I upgraded the system from 10.3E to 11.5E using this [guide](https://3ds.guide/homebrew-launcher-(soundhax)). Every other game I own works without issue, but this one crashes. Before the upgrade it worked :)

During the upgrade, I got a black screen with luma that I could fix by deleting `/Nintendo 3DS/(32 Character ID)/(32 Character ID)/extdata/00000000/`. In this ticket I stumbled across this [wiki entry](https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-troubleshooting) and as a precaution I applied those steps (since, I already run CFW for quite a while).

---

After that the upgrade finished without issues. Thus, trying out different games I own.

Then starting Pokemon X, I got this error: 

[image]

I could fix this error by deleting `/Nintendo 3DS/(32 Character ID)/(32 Character ID)/title/0004000e/00055d00/`. But, then I got this new one (dump is attached)

[image]

**Steps to reproduce:**

1. Insert Pokemon and start it
2. It crashes

**Dump file:**

[crash_dump_00000018.zip](https://github.com/AuroraWright/Luma3DS/files/1246973/crash_dump_00000018.zip)

Sonlen1414, 2017-08-23:
The error in the first picture is a File System check fail, most likely the update data is corrupted.

Hikari-chin, 2017-08-24:
Pokémon X works fine for me. This is typically damaged update/game data. Try deleting the update data in System Settings>Data Management and redownloading it.

closed 2017-08-24 by Hikari-chin

rds12, 2017-08-24:
@Hikari-chin I already deleted the files in `System Settings>Data Management`. After that I'm in the state depicted in my second picture. A redownload brings me into the state depicted in my second picture.

When I try to access the "Manual" from the main menu. I get

> An error occurred (ErrDisp)
Error type: card removed
The card was removed.

So what other options do I have?