LumaTeam/Luma3DS #816: An exceptation occured error ARM9 while booting on O3ds (prefetch abort) (svcBreak)

PaperApple, 2017-08-24:
**System model:**

[Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[11.2.0.35U]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap/Sighax]

**Luma3DS version:**

[ 8.1.1 stable]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
This 3ds was bricked not to long ago and I unbricked it with NTRboot Hax. I followed all the steps properly, but everytime I reboot it goes to a screen with a bunch of codes I wana know if there is anyway to fix this issue. If so please tell me I been trying for hours and found no solution. 

**Steps to reproduce:**

1.

2.

**Dump file:**

[image]

[Uploading dumps.zip…]()

Hikari-chin, 2017-08-24:
Not a Luma issue; likely something wrong with your "unbricked" NAND.

closed 2017-08-24 by Hikari-chin

PaperApple, 2017-08-24:
Is there a fix?

Derek1199, 2017-08-24:
Try doing a CTRNAND transfer of a clean Nand image. That should fix it.

PaperApple, 2017-08-24:
How do I do that?

HugaShire, 2017-08-25:
Not a support form. Look at 3DS.guide/9.2.0-ctrtransfer