LumaTeam/Luma3DS #818: System Settings causes Crash

strife98, 2017-08-24:
**System model:**

N3ds 

**SysNAND version (+emu/redNAND version if applicable):**

11.3 or 11.4. I can't get into system settings to check. 

**Entrypoint (How/what you're using to boot Luma3DS):**

godmode9, boot9strap 

**Luma3DS version:**

8.1.1-f6483ec

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
The owner of the 3ds tried to do a system transfer to put CFW on another 3ds. Instead of waiting the week to transfer back her NNID, she did did a SysNAND restore. Since then, going into system settings causes the 3ds to crash with the ARM11 crash. I have tried to get rid of the NNID with Godmode9, ctrnand transfer, and tinyformat which crashed when it restarted. I have tried posting to gbatemp, and the Nintendo Homebrew Discord to try to fix this, but they suggest one thing and then when I say it didn't work, they stop saying anything. I know this isn't a support forum, but I'm running out of support options. I'm hoping the crash report will explain how to fix it. 
**Steps to reproduce:**

Unadvisable honestly, but:
1. System Transfer
2. Restore backed up SysNAND
3. Go to system settings

**Dump file:**
[crash_dump_00000014.zip](https://github.com/AuroraWright/Luma3DS/files/1250254/crash_dump_00000014.zip)

HugaShire, 2017-08-25:
Reinstalling System Settings. This happens alot. Not a luma issue

strife98, 2017-08-25:
How would I go about doing that? Do I just look for a System Settings Cia or something?

HugaShire, 2017-08-25:
Yes or update the system in Recovery mode

closed 2017-08-25 by Hikari-chin