LumaTeam/Luma3DS #820: Arm11 an exception occured with Games and Homebrew.

ooLIAMoo, 2017-08-26:
**System model:**
Old 3DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
9.2.0-20E SysNAND
[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap 1.2
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
v8.1.1 Stable
[e.g. 8.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Attempting to launch the downloaded Pokémon Sun game results in a ARM11 exception occurred screen which doesn't allow me to boot into the game. (Redownloading game has caused a never-ending black screen with no sound.)
Also, attempting to boot into homebrew such as 'eShop' (that comes with the homebrew starter zip) also gets me a similar ARM11 exception occurred error.

These error codes were gotten from the most recent nightly version, however the error still happens even if I use a stable build - like I'm currently on.

Game error:
Processor: ARM11 (core 1)
Exception type: prefetch abort (svcBreak)
Fault status: Debug event
Current process: loader (0000000000000000)

Homebrew error:
Processor: ARM11 (core 0)
Exception type: data abort
Fault status: Translation - Section
Current Process: 3dsx_app (000400000D921E00)

**Steps to reproduce:**

1. No idea. I followed the 3ds.guide setup completely without missing anything.

2.

**Dump file:**

I removed most of the dump files (or they removed themselves after I changed to stable build) so I only have this dump file inside of the folder.
[dump.zip](https://github.com/AuroraWright/Luma3DS/files/1253356/dump.zip)

Hikari-chin, 2017-08-26:
Pertaining to Pokémon Sun, that behavior is caused by the incompetence of a certain piracy application, is not a Luma issue, and will not be supported here. As for the starter kit, much of it hasn't been updated for Rosalina and there's no reason to use the deprecated "eShop" update check bypass 3DSX anymore either.

closed 2017-08-26 by Hikari-chin

ooLIAMoo, 2017-08-26:
can I get an exact reason? such as CIA files downloaded from CIAngel not working and having to download from somewhere else or something?