LumaTeam/Luma3DS #821: Switch Homebrew exploit from Rosalina to Hax 2.5?

SplatMan333, 2017-08-26:
Hi, I was wondering how I could switch my homebrew exploit from Rosalina to Hax 2.5. Thank you!

Shadow322, 2017-08-26:
Why?

MarioMasta64, 2017-08-26:
theres a few things:
i cant remember all of them but heres one i can remember:
being able to select titles. in rosalina this is not a feature.
im sure we can make this work. rosalina can be good. and i believe it will be.

ihaveamac, 2017-08-26:
\*hax still won't work, because: https://github.com/AuroraWright/Luma3DS/issues/495#issuecomment-306149488

> hax payloads by design cannot work under Rosalina because the memory map is shuffled around to make space for the Rosalina system module to load (and this breaks the hax payloads because they expect certain things to be at fixed locations in memory).

> It was decided that target title selection is not useful anymore since 3dsx applications have full ARM11 system permissions now, and it is possible to directly access the savefiles of any title by using the appropriate FSUSER commands. In the meantime, you can use the CIA versions of savegame managers.

closed 2017-08-26 by ihaveamac