LumaTeam/Luma3DS #824: ARM11 exception, type: data abort

conanac, 2017-08-27:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

9.2.0-20U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

8.1.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Using ntrboot method to install boot9strap (following 3ds.guide).
Deleting boot.firm (from renaming file SafeB9SInstaller.firm) in the root of SD card and then copying boot.firm from the Luma3DS. 
Booting into the Luma3DS configuration menu and turning on “Show NAND or user string in System Settings”.
Pressing (Start) to save and reboot, then after rebooting ARM11 exception error happens.
Every rebooting produces ARM11 exception error (and stuck there).
Only uninstall CFW script running with GodMode9 helped me get this old 3DS to show 3DS home menu again (i.e. by uninstalling boot9strap hax).

**Steps to reproduce:**

1. Rebooting always produces ARM11 exception error.

2.

**Dump file:**

[crash_dump_00000001.zip](https://github.com/AuroraWright/Luma3DS/files/1254115/crash_dump_00000001.zip)

FrozenChen, 2017-08-27:
This is issue is fixed by deleting your home menu extdata. Its located in 
/Nintendo 3DS/(32 Character ID)/(32 Character ID)/extdata/00000000/ ,delete the 0000008f folder located there

mzino, 2017-08-27:
Should be the same problem I had just yesterday.
https://github.com/AuroraWright/Luma3DS/issues/822

I solved by deleting extdata as FrozenChen said (to be sure, delete extada for both SysNAND and EmuNAND if you have one).

conanac, 2017-08-27:
Thanks, it works.
Now I remember in the older version of guide, there was a step to delete that folder, but somehow it is not in the new one. Perhaps if it is not there, it should be added back to the troubleshooting part of the guide?
I close this issue now.

closed 2017-08-27 by conanac

HugaShire, 2017-08-29:
@conanac Its still in the troubleshooting area, for black screen on boot. Also this was only on the guide because the 2.1 CTRtransfer wouldn't get the 3DS to boot since it cant properly read the data because it isn't for that version.

aprilsz, 2019-05-06:
can anyone help. ive encountered the same issue, as i am trying to set up ledybot

skullboy64, 2020-05-04:
I have multiple 0000008f folders do i delete them all

Hoffmannn, 2021-02-09:
Doing this has corrupted my SD card, DON'T do this.

> 
> 
> This is issue is fixed by deleting your home menu extdata. Its located in
> /Nintendo 3DS/(32 Character ID)/(32 Character ID)/extdata/00000000/ ,delete the 0000008f folder located there

urherenow, 2021-02-09:
No, doing that does not corrupt anything. Your own actions or faulty sd card did. Like pulling it out of your PC without "ejecting" it first, or out of your 3ds without powering off first.

Slashin-it-up, 2021-02-09:
> Doing this has corrupted my SD card, DON'T do this.
> 
> > This is issue is fixed by deleting your home menu extdata. Its located in
> > /Nintendo 3DS/(32 Character ID)/(32 Character ID)/extdata/00000000/ ,delete the 0000008f folder located there

Deleting exdata does nothing else then deleting the home menu data, which is the folder and placement of icons

Hoffmannn, 2021-02-09:
> 
> 
> No, doing that does not corrupt anything. Your own actions or faulty sd card did. Like pulling it out of your PC without "ejecting" it first, or out of your 3ds without powering off first.

> Deleting exdata does nothing else then deleting the home menu data, which is the folder and placement of icons

Weird, I didn't do anything different then what I always do. It was only when I deleted exdata that my 2DS could no longer read the SD card, and after that I couldn't read it on my PC either. Maybe it was bad luck then, because I wasn't careless.

JUN-SUZU, 2021-03-22:
0000008fフォルダは見つかりませんでした。

kevsestrella, 2021-04-27:
> 0000008fフォルダは見つかりませんでした。

00000082 for JPN Region 3ds

toxinu, 2021-11-11:
00000098 for EU Region 3ds