LumaTeam/Luma3DS #826: ARM 11 Exception, Data Abort

Manurocker95, 2017-08-27:
[arm11.zip](https://github.com/AuroraWright/Luma3DS/files/1254322/arm11.zip)

**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.5.35 Eur Sysnand

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2 (Latest one)

**Luma3DS version:**

8.1.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I'm making a homebrew for watching anime and whenever I set a new variable on my game (the one that runs the core of the homebrew), it breaks. If I set it locally it works without issue. It's really weird. 

Source code here:
[https://github.com/Manurocker95/Anim3DS](url)

**Steps to reproduce:**

1. In game.h just create a new variable or two. It doesn't matter what it is. I tried always integer, so: int var, var2;  

2. Compile and create the .cia. 

**Dump file:**

I attach some of my dumps (Should be the latest ones)

ihaveamac, 2017-08-27:
this is very likely an issue with your code, not luma. luma is just telling you the exception details.

closed 2017-08-27 by ihaveamac

Manurocker95, 2017-08-28:
But I dunno why it happens as it's the only thing giving me issues

ihaveamac, 2017-08-28:
what do you mean?