LumaTeam/Luma3DS #828: Bug Pokemon Lune for old 3ds

polopoi, 2017-08-28:
I play Pokemon Moon and it bug in the volcanic park when I try to face other similar trainers for the test of Kiwae here is the message that I receive:

An error occurred (ErrDisp)

Error type: generic
Process ID: 38
Process name: niji_loo
Process title ID: 0x0004000000175e00

Adress: 0x002b37401
Error code: 0xo96045b 

Press any button to reboot

help me 
i use sysnand no emunad my system is 11.5.0-38E ,my luma 3ds 8.1,FBI is 2.4.11

AuroraWright, 2017-08-28:
The error code is missing two digits

polopoi, 2017-08-28:
Repair an Error Code? How do we do ? I'm new to the crack of the 3ds
It would be good if you showed me the trick how to repair this erratic ca will be able to serve me for the other games that I have on my console.
What more do you mean by the error code missing two digits? I would like to have more details on this error.

Thanks you for your comprehension

Committee, 2017-08-28:
What she means is the error code that you gave is missing two characters:

Error code: 0xo96045b??

What are the missing characters?

polopoi, 2017-08-29:
ok i see , this is the true error code : 0xo960454b

Committee, 2017-08-29:
You posted the same thing.

HugaShire, 2017-08-29:
This could be an issue with many things. However you error code you keep telling us is missing a digit

polopoi, 2017-08-29:
look the picture 

-
[image]

fincs, 2017-08-29:
Turns out it was a "C", not an "O".
```
<fincs> !err 0xc960454b
<plutobot> Module: FS (17) | Description: 331 | Summary: Internal (11) | Level: Status (25)
```

polopoi, 2017-08-29:
that's it sorry for the bad writing error , i need a solution for resolve this error

Hikari-chin, 2017-08-29:
Can't reproduce this; this is probably game or update corruption.

closed 2017-08-29 by Hikari-chin

polopoi, 2017-08-30:
how to repair this error ?

james-d-elliott, 2017-08-31:
You can get support for issues not related to Luma on the discord server.  https://discord.gg/C29hYvh

polopoi, 2017-08-31:
In which part of the site I have to go to see how to solve this problem that prevents me in my adventure in pokemon moon, I invite you to see a little higher I posted the image of the error