LumaTeam/Luma3DS #829: Add support for DLC patching

mariomadproductions, 2017-08-28:
Suggestion: Add support for patching DLC titles. DLCs have multiple "app" files, therefore multiple RomFS sections and multiple code binaries. Luma3DS does not seem to have support for this from reading the wiki.

Hikari-chin, 2017-08-29:
This is likely unfeasible; the mounting locations are unknown and are probably game-specific. If not, this might be possible.

jonirrings, 2018-07-26:
@mariomadproductions 
yes, it can't be autodetected by looking at the game's code.
cause every game has its own binary layout/mapping, and it own logic to decode the mapping.

that is the difficulty what the community translation groups had encountered.
every game requires manual reverse engineering, then replace the text and assets~~~

mariomadproductions, 2018-07-26:
Could it be autodetected by looking at what locations the game tries to use while it is running?

jonirrings, 2018-07-27:
logically possible, but that requires much much work.
a program can't distinguish whether the game is fetching a code or an asset, because it is game-specific, it will vary every time. the code in one game may look like an asset in another one. 
so for "auto-detect", you should specify rules for the program to judge, even for a HACKER, what she/he did is "guess and try", based on her/his experience.

as said above, every game has its own binary layout/mapping, without this, code looks the same as asset, and you can't tell the border of them. to get this, you have to analyse every game manually. that is why it is hard to make "auto-detect".

mariomadproductions, 2018-08-03:
I'm not really following. The code is in a separate file to the assets: ``.code`` in the ExeFS. Also, won't the code use some kind of service to get the DLC data? And doesn't DLC always have some kind of ``SOMETHING://``string the ``.code`` like ``BIGRED://`` in NSMB2?

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH