LumaTeam/Luma3DS #831: Rosalina reboot - HOME menu ignores latest focused tile

DMSalesman, 2017-08-30:
**System model:**
Old 3DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.5.0-38E SysNAND
[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap/Sighax
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.1.1
[e.g. 8.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (1)

Splash: (Before)

PIN lock: (Off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
If shutting down the 3DS with either the hardware button or the `Power off` setting inside Rosalina, the HOME menu will display the last focused tile on the next boot.

If rebooting the device by means of Rosalina's `Reboot` setting, though, the HOME menu won't display the last focused tile upon reboot, showing the one before the last instead.

It would appear to me that rebooting via Rosalina doesn't let the HOME menu save the last tile, but I'm unsure whether this is either desired behavior or non-fixable.

**Steps to reproduce:**

1. When inside the HOME menu, note down the currently focused tile (for example, **System Settings**).

2. Power off the 3DS either by using the hardware button or with Rosalina's `Power off` setting.

3. Boot the 3DS, and note that the HOME menu displays the same focused tile noted down on step 1 (i.e. **System Settings**).

4. Change your focus to another tile (for example, **Nintendo 3DS Sound**).

5. Reboot the handheld via Rosalina's `Reboot` setting.

6. Note that the HOME menu still displays the focused tile noted down on step 1 (i.e. **System Settings**), while we expected it to focus on the one noted down on step 4 (i.e. **Nintendo 3DS Sound**).

**Dump file:**

SvenDaHacker64, 2017-08-30:
It happens to me. HOME Menu can only save focused title when going to Change Theme menu, Manual, applets (Friend List, Internet Browser), etc...
That's why Rebooting is so fast enough for Home Menu to have a chance to save which is too late (just like interrupting the game save progress by pushing power button), while Power off is taking a while to completely shutdown and Home Menu can still save.

DMSalesman, 2017-08-30:
Indeed. That's why I mentioned that this may be the Reboot feature's intended behavior.

james-d-elliott, 2017-08-31:
https://github.com/AuroraWright/Luma3DS/blob/master/sysmodules/rosalina/source/menus.c#L92-L143

Think it's to do with the srvPublishToSubscriber() method and svcKernelSetState() methods. I doubt that there is a notification ID for a reboot so they used a kernel method instead which skips the normal shutdown process. It could probably be researched and a fix found but I doubt it's worth the effort.

DMSalesman, 2017-08-31:
I agree that this is a minor issue, but I thought fair to bring it up, since the rest of the firmware is rock solid and seems to have been designed by excellently perfectionist people.

Now, if they think it worthy to be fixed, it's fine. If not, it's fine either way! As said, this is not something that breaks the experience, and therefore can be looked into when far more urgent problems have been addressed.

profi200, 2017-08-31:
This may be because it uses svcKernelSetState() which is not a clean reboot. I would make the reboot option "leave" (so all threads continue to run) the menu and use the APT function for reboot/poweroff instead.

LiquidFenrir, 2017-09-30:
The same problem (not clean reboot) causes the camera led to stay on if you reboot with rosalina while in the camera applet/application, but I've been told this was a wontfix/very low priority.

DMSalesman, 2017-09-30:
@LiquidFenrir

I see. Personally, I'm no avid user of the Reboot feature, and I happened to stumble on this by chance only.

commit 9c5766f6 (2017-11-24): Merge pull request #945 from astronautlevel2/master

Switch to APT_ResetHardwareAsync() for reboot (Closes #831), fix #831
commit c9701f93 (2017-11-24): Switch to APT_ResetHardwareAsync() for reboot (Closes #831)
closed 2017-11-24 by Hikari-chin