LumaTeam/Luma3DS #832: [Feature Request] Custom LED colors

BlazeUchiha420, 2017-08-30:
Is it possible to have an option in Rosalina to customize the color of your LED lights on your console similar to what is shown in this video here: https://www.youtube.com/watch?v=i9ILMxlcUik without the possibility of an unrecoverable MCU brick? If so, then it would be awesome to have an option for every LED on the console. This would allow for more customization for the end user's preference(even the potential to have have them continuously change like in the video.

Derek1199, 2017-08-31:
#752 go here.

BlazeUchiha420, 2017-08-31:
sorry for posting in the wrong area. Thanks for rectifying this for me!

closed 2017-08-31 by Hikari-chin

BlazeUchiha420, 2017-08-31:
Not to sound entitled or anything, may I ask why this is invalid and closed? Is it not possible?

ihaveamac, 2017-08-31:
it's closed since these kinds of things go to the feature-request issue.

BlazeUchiha420, 2017-09-01:
So would it be ok if I re-posted it there or would it just get closed again? Afaik it was moved there.

Hikari-chin, 2017-09-01:
It would just be deleted for being a duplicate; this feature was already requested.

BlazeUchiha420, 2017-09-01:
alright thank you