LumaTeam/Luma3DS #834: 3 to 4 launches of the Debugger while offline crashes the 3DS

killerfurze, 2017-08-31:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9strap 1.2

**Luma3DS version:**

8.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: Clock+L2 (x)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: (x)

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
I was offline and pressing buttons when I was bored in the car. I launched Debugger 2 times and got the normal error from it. The Third time I launched it the 3DS crashed. If I remember I had to press something else.

**Steps to reproduce:**

1. Go offline

2. Keep enabling the debugger

**Dump file:**

[crash_dump_00000001.zip](https://github.com/AuroraWright/Luma3DS/files/1268216/crash_dump_00000001.zip)

SvenDaHacker64, 2017-09-06:
This happened to me.

T3CHNOLOG1C, 2017-09-27:
Can reproduce on Old 3DS 11.6.0-39U, Luma v8.1.1-f27cdb45

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH