LumaTeam/Luma3DS #837: Crashing launching DosBox

Waces, 2017-09-01:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

https://github.com/AuroraWright/Luma3DS/commit/bc1aa15dd709a703b9af1a0f89ccded21fc08823

**Luma3DS configuration/options:**

Default EmuNAND: ( 1)

Screen brightness: (4 )

Splash: (Before )

PIN lock: (Off )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X )

Show NAND or user string in System Settings: ( X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Whenever trying to launch DosBox (the CIA) it just crashes.
With 'Exception Type: data abort, and Fault status: Translation - Section'

**Steps to reproduce:**

1. Install DosBox from CIA or Homebr3w

2. Open DosBox

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1271049/crash_dump_00000000.zip)

Hikari-chin, 2017-09-02:
Did some research on this; looks like DOSbox for 3DS is quite old and unmaintained; this is very likely the cause of these issues, not Luma.

closed 2017-09-02 by Hikari-chin

PJkeeh, 2018-04-26:
Got the same issue sadly. I guess you never found a solution

Robz8, 2018-04-26:
It's not a Luma issue. The dev of DOSbox for 3DS will have to fix it.