LumaTeam/Luma3DS #838: feature request: title region emulation automatically based on title serial

Jimmy-Z, 2017-09-01:
I requested this in an "all feature requests goes here" thingy, but I can't find it now, so posting it again.

Currently users need to create a text config file to engage title region emulation, but since region of commercial games can be determined based on the last letter of the serial, we can set title region emulation to that if it doesn't match system region, eliminating the requirements of such text config files.

Hikari-chin, 2017-09-02:
#619 was also only for Rosalina feature requests, as the title said. Unfortunately, this feature, while possible, will not be implemented, for the following reasons: 

* The SMDH region is not easily accessible by something like Loader; to check it every launch would incur a performance penalty.
* The standard is not necessarily followed by unofficial applications. 
* General buggy behavior this could cause.

closed 2017-09-02 by Hikari-chin

Jimmy-Z, 2017-09-02:
performance penalty like several milliseconds?

give me one example of unofficial application that will fail with title region emulation on.

how exactly?

that locale switcher is a joke.

piratesephiroth, 2017-09-02:
well it could skip the check for system titles. I don't really see a strong impact in performance if you do that