LumaTeam/Luma3DS #840: i can not access the homebrew after i installed luna

biggiedaman, 2017-09-04:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

biggiedaman, 2017-09-04:
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1273542/crash_dump_00000000.zip)

Shadow322, 2017-09-04:
Should fill out the spreadsheet properly if u want any help

biggiedaman, 2017-09-04:
i am trying. i am new at this. i will post a pic of what i get when i try to load homebrew from soundhax

killerfurze, 2017-09-04:
Luma breaks some homebrew entry points. That might be the issue.

biggiedaman, 2017-09-04:
i solved it thank you for your time and help every one. i just had to inject the fbi into the health and safety by using godmode9.

fincs, 2017-09-04:
*hax 2.x is not supported by Luma due to differences in RAM offsets. Please note that Luma has a built-in way to run 3dsx homebrew applications.

closed 2017-09-04 by fincs

Jenarou, 2017-09-11:
can you please explain how did you fix it?
I am having the same issue and I have no idea how to fix it using godmode9, I have godmode9 installed already.

biggiedaman, 2017-09-11:
Just google godemode9 usage