LumaTeam/Luma3DS #841: Screen flickering after crash + how to fix

DatRobinPlays, 2017-09-05:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-*E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

 Boot9Strap

**Luma3DS version:**

8.1.1-f6483ec

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

If the Crash Report is active and my n3dsxlE crashes, screen sometimes starts flickering after restarting then. Since I turned the crash report off, it wasn't flickering again after crashes. 

**Steps to reproduce:**

1. 

2.

**Dump file:**
Sorry, but I don't want to enable the crash thing right now, like you may understand.

**
Sorry for the less informations but I'm not at home atm and I wanted to report before I forget it again. For questions, ask please

Hikari-chin, 2017-09-05:
This has already been fixed. Thank you.

closed 2017-09-05 by Hikari-chin

DatRobinPlays, 2017-09-10:
Turned the Arm 11 Exception Handler back on. After a crash after a few days again flickering and this crazy stuff - never seen before. I don't know how to upload images here so here's a folder with them. http://DatRobinPlays.de/owncloud/s/lUEhcfDjvA45YbV

ihaveamac, 2017-09-10:
did you test the latest commit?

DatRobinPlays, 2017-09-10:
Well. I think I'm to stupid for that at the moment. I don't know about GitHub and it's about 4 at night right here. Also I don't know what a commit is. Google translate says it is something like a must do. 

Pls don't angry with me :c

Derek1199, 2017-09-11:
@DatRobinPlays Latest commit is basically the latest code changes to the project. That's what @ihaveamac is meaning in this case. This issue has been fixed already in the latest commit.