LumaTeam/Luma3DS #842: Freeshop: either crash->Home Menu or Error Screen (see image below)

c12345678901, 2017-09-06:
[New 3DS (XL)]

**SysNAND version (+emu/redNAND version if applicable):**

[11.4.0-37E SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap/Sighax.]

**Luma3DS version:**

[8.1.1 Luma]

**Luma3DS configuration/options:**

Default EmuNAND: 1(x)

Screen brightness: 3(x)

Splash: Off(x)

PIN lock: Off(x)

New 3DS CPU: Clock(x)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Hey there, about two weeks ago I wanted to open FreeShop and it either crashed right away or it loaded a few things and then it showed me an error screen "An exception occurred"
I also tried to reinstall the app or deleting the freeshop folder but that was useless.

 

**Steps to reproduce:**

1. opening freeshop
[crash_dump_00000010.zip](https://github.com/AuroraWright/Luma3DS/files/1282040/crash_dump_00000010.zip)
[image]

2.

**Dump file:**

c12345678901, 2017-09-06:
[11.5.0-38E SysNAND]* sorry

AuroraWright, 2017-09-06:
1) not Luma's issue
2) we don't support piracy

closed 2017-09-06 by AuroraWright

pbanj, 2017-09-06:
simple fix....... buy your games