LumaTeam/Luma3DS #843: {Feature Request} Screenshot placement

Shadow322, 2017-09-09:
When Luma is in ctrNAND taking screenshots with Rosalina it places all screenshots in "RW/Luma/Screenshots" have it save screenshots in "SD:/Luma/Screenshots"

Derek1199, 2017-09-09:
It's supposed to save them in Sysnand CTRNAND/rw/luma/screenshots when you are booting without an SD card. This isn't a luma issue at all. It can't save to an SD card when no SD is present. If booting with an SD card it will save To SD/luma/screenshots

Shadow322, 2017-09-09:
Well your wrong as the SD card is present and still saves to "CTRNAND/rw/luma/screenshots" so it is a luma issue

jupitteer, 2017-09-09:
@Shadow322 Just put boot.firm on your sd card.

Shadow322, 2017-09-09:
@jupitteer just put your brain in a jar of fermadahide as its doing u no good

closed 2017-09-09 by Hikari-chin