LumaTeam/Luma3DS #844: File is corrupted

Jenarou, 2017-09-11:
Hi,  the luma8.1.7z file is corrupted.

Thanks.

Shadow322, 2017-09-11:
Then redownload it thanks

Jenarou, 2017-09-11:
I am following the guide for 3ds hacking and I downloaded all files except this one,   the file is corrupted no matter how many times i download it .... try to download it and see yourself.

Hikari-chin, 2017-09-12:
File is not corrupted, tested.

closed 2017-09-12 by Hikari-chin

ihaveamac, 2017-09-12:
both 8.1 and 8.1.1 downloads verified here.

Jenarou, 2017-09-12:
ihaveamac , thank you 
this is what i exactly get when i try to open the file, note its the only 7z file that do this.

anyway thanks guys ill try to download it on another OS and see how it goes maybe its my system.

cheers many thanks for the help.

[image]

HugaShire, 2017-09-14:
Use 7zip or Unrar.