LumaTeam/Luma3DS #847: [Bug] LayeredFS Speed on New 3DS

sora10pls, 2017-09-13:
I've done several tests, and when comparing my two systems with the same SD Card (Old 3DS has a microSD inside an adapter) and the same contents on the SD Card, LayeredFS is seemingly faster on the Old 3DS than it is on my New 3DS.

I'm not sure why this is even the case, as I do have the New 3DS CPU set to Clock+L2 in my n3DS' Luma Config, so...

I haven't seen anybody else report on this but the games I tested with were Pokemon Y and Pokemon Alpha Sapphire. When loading menus in a battle, the load times are extremely noticeable on New 3DS but are almost the exact same to vanilla load times on my Old 3DS...

If anybody with both models could test as well, that'd be great. Not sure if it's a "just me" thing.

ValliantStorme, 2017-09-13:
Have you tested with Clock+L2 disabled (and permutations of the two)? It would be interesting if it were an issue with either Clock or L2 cache

sora10pls, 2017-09-13:
Still very slow regardless.

AuroraWright, 2017-09-20:
LayeredFS just uses FS commands to open files. So unless Nintendo screwed up FS on N3DS (unlikely) this is due to the random access speed/latency difference between your SD and your microSD cards.

closed 2017-09-20 by AuroraWright