LumaTeam/Luma3DS #848: Cannot find the boot.firm file when I download the latest luma3DS its just the version .7z as one file? Am I just doing it wrong?

doopeeperdoodle, 2017-09-14:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

astronautlevel2, 2017-09-14:
Not entirely sure what's going on here;

If it's the latest release (what most people are using, https://github.com/AuroraWright/Luma3DS/releases) you need to extract the 7-zip archive. You can download 7zip, which is an open source archive program, from here: http://www.7-zip.org/download.html

If it's the latest commit, you need to either download the source and build it or download a precompiled source from my site (https://astronautlevel2.github.io/Luma3DS) and extract the zip file and grab the boot.firm

closed 2017-09-14 by Hikari-chin

Derek1199, 2017-09-14:
You can also use Winrar.

astronautlevel2, 2017-09-14:
Please don't recommend closed source, outdated tools here.